ros/roscpp_core #35: No NodeHandle::getParam() overload for float?

theidealist, 2015-06-03:
I'm not sure if this is the correct project, but it seems like the correctly named one and I could not find which repo actually has node_handle.h.

In my version of ros (indigo), NodeHandle does not have a ::getParam() overload for float, the rationale for which is not clear.  This seems like a bug/oversight.

Thank you!

dirk-thomas, 2015-06-03:
Can you please open a ticket in the following repository which contains the NodeHandle class: https://github.com/ros/ros_comm and then close this one.

dirk-thomas, 2015-06-03:
The parameter server only stores double values - never float values. Therefore the `NodeHandle` API only exposes `getParam()` with `float` and leaves the conversion (with potential precision loss) to the user.

The choice for not storing floats natively on the parameter server is related to cross language support: http://answers.ros.org/question/10938/why-cant-you-use-floats-for-accessing-parameters-in-roscpp/

Based on that I don't think the function is missing is not an oversight but intended.

theidealist, 2015-06-03:
Thanks @dirk-thomas, I'll move it there for confirmation you're right (sounds plausible anyway)

closed 2015-06-03 by theidealist