ros/roscpp_core #39: WallRate and Rate have have different Sleep-Behaviour

NikolasE, 2015-11-02:
According to the documentation, ros::Rate should return false if the desired rate was not reached (https://github.com/ros/roscpp_core/pull/37 will make sure that is is also implemented). 

However, the docu of WallRate.sleep() says that it will return true if the rate was not reached which is surprising. I think that both classes should behave the same so that the docu (and implementation) of WallRate should be aligned.

dirk-thomas, 2016-03-01:
#45 fixes the return value of `Rate::sleep`.

`WallRate::sleep` will be updated in K-turtle - the documentation as well as the implementation - to be similar to `Rate::sleep` and allow to distinguish if the function slept as expected or not.

dirk-thomas, 2016-03-17:
#47 changes the semantic of `Duration::sleep` to match the return value semantic of `Rate::sleep`.

@NikolasE Can you please give the branch of that PR a try to confirm that it behaves in the way you expect it to.

closed 2016-04-12 by dirk-thomas