ros/roscpp_core #42: Fix Serializer<bool> on ARM

tho-, 2016-02-03:
Here is a patch fixing an issue with bool serialization on ARM.

[bool-serialization.diff.txt](https://github.com/ros/roscpp_core/files/115474/bool-serialization.diff.txt)

dirk-thomas, 2016-02-23:
With a short test I was able to send messages containing boolean fields from ARM to non-ARM without any problem.

Please provide more information about:
* what exactly you tested
* what the expected and actual behavior was
* a [sscce](http://sscce.org) to reproduce the problem

Also instead of providing a diff file please consider creating a pull request.

tho-, 2016-02-23:
I though it was so simple that I didn't want to bother you with details. "sizeof(1)" obviously makes no sense here, it looks more like a typo than a bug. Anyway, here you go:

On x86_64:
```
cricket[~] > cat test-bool-serializer.cc
#include "ros/serialization.h"

int
main()
{
  uint8_t c[1];
  ros::serialization::Serializer<bool> s;
  ros::serialization::OStream o(c, sizeof(c));

  s.write(o, true);
  return 0;
}
cricket[~] > uname -m
x86_64
cricket[~] > source /opt/ros/indigo/setup.zsh
cricket[~] > g++ test-bool-serializer.cc -L/opt/ros/indigo/lib -lroscpp_serialization
cricket[~] > ./a.out
cricket[~] > echo $?
0
```

While on arm:
```
alexanor[~] > uname -m
armv7l
cricket[~] > source /opt/ros/indigo/setup.zsh
cricket[~]!> g++ test-bool-serializer.cc -L/opt/ros/indigo/lib -lroscpp_serialization
cricket[~] > ./a.out
terminate called after throwing an instance of 'ros::serialization::StreamOverrunException'
  what():  Buffer Overrun
zsh: abort (core dumped)  ./a.out
alexanor[~]!> echo $?
134
```

tho-, 2016-02-23:
If you prefer, you can also pull the commit from there:
`git pull --rebase http://homepages.laas.fr/mallet/roscpp_core.git indigo-devel`

dirk-thomas, 2016-02-25:
Thank you for providing the example snippet. The problem doesn't happen when using message serialization and therefore I could reproduce this without. Only in the above example which uses a custom buffer (which doesn't have more bytes to accommodate the extra three bytes) I was able to reproduce the problem and also confirm that the patch you recommended addresses the problem.

PR #44 (2016-02-25): fixed serializer bool on ARM
Fixes #42.

Without the fix message serialization still works since it just happens to write three extra bytes which it shouldn't. The boolean value is correctly represented in the byte stream. But with a known allocated storage it might write over the expected limit.

The problem especially happens when using a serializer instance with a custom buffer which doesn't have the three extra byte at the end as in the reference ticket.
closed 2016-02-25 by dirk-thomas