ros/roscpp_core #54: Warning for unused indent parameters in message_operations.h

amiller27, 2017-02-09:
PR #52 which dealt with this issue on clang breaks on gcc 6 (gcc doesn't recognize the void cast as a valid use of the variable and still emits the warning).

Is there a reason the indent is unused in the first place?  That is, why aren't these
```
s << indent << thing_to_output << "\n";
```

dirk-thomas, 2017-02-10:
@mikepurvis Can you maybe take a look at this since you contributed #52.

@amiller27 Assuming that it is intentional that the argument is not used can you please suggest an alternative which works with GCC 6. For documentation purposes we certainly don't want to remove the variable name from the signature itself.

amiller27, 2017-02-10:
Sorry about this, just checked again and it's working, gcc was using the wrong header and didn't have the cast to void.  My bad.

closed 2017-02-10 by amiller27