ros/roscpp_core #75: Request rebuild of cpp_common for melodic

mikaelarguedas, 2018-03-03:
Since the last bionic package import yesterday, the `cpp_common` package cannot be installed because its' dependencies are not available anymore (libconsole-bridge is now 0.4 and not 0.2).

apt console output below:
<details>

```
root@532af06e867b:/# apt install ros-melodic-cpp-common
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-cpp-common : Depends: libconsole-bridge0.2v5 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

dirk-thomas, 2018-03-03:
If it doesn't need a new release or deb inc but only a rebuild of the job can you trigger the Jenkins job in question?

mikaelarguedas, 2018-03-03:
Yes, I just wanted a public trail of it. I retriggered amd64, once all downstream jobs pass I'll trigger the arm jobs and close this issue

mikaelarguedas, 2018-03-05:
Rebuild completed and now all packages are installable on amd64 (arm rebuild pending). Closing this

closed 2018-03-05 by mikaelarguedas