ros/roscpp_core #77: duration conversion bug in OSX

BrutusTT, 2018-03-20:
Doing a source-based installation I came across a failing test in rostime on OSX which looks similar to #48  / #50 but for the Duration class (TestCase: Duration.ToFromSec). However, it seems it only appears on  OSX the test passes on Ubuntu not 100% sure what the difference exactly is.

The following change prevents it:

rostime/include/ros/impl/duration.h line 59

original:
  `nsec = (int32_t)((d - (double)sec)*1000000000);`

modified to:
   `nsec = (int32_t)((d - (double)sec)*10000000000/10);`

Idea: There seems to be problem in bitwise representation. By shifting the whole double by a factor of 10 it seems to be gone and the backwards division by 10 repairs for it.

OSX:    10.11.6
LLVM:   version 8.0.0 (clang-800.0.42.1) 
Target: x86_64-apple-darwin15.6.0)

ProfFan, 2018-03-23:
Investigating a similar issue where rostopic hz shows 1567 Hz in OSX with a 1000 Hz message.

dirk-thomas, 2018-04-02:
@ProfFan can you please try the referenced patch and comment if that addresses your problem. If not it might be unrelated and it would be better to fill a separate issue with sufficient information to reproduce your problem.

dirk-thomas, 2018-04-02:
Addressed by #78.

closed 2018-04-02 by dirk-thomas