ros/roscpp_core #84: No matching conversion from 'double' to 'pt::microseconds'

mbreyer, 2018-05-28:
When compiling the rostime package on my machine, I get the following compiler error:

```
/Users/michel/Developer/kinetic_ws/src/roscpp_core/rostime/include/ros/impl/time.h:170:35: error: no matching conversion for functional-style cast from 'double' to 'pt::microseconds' (aka 'subsecond_duration<boost::posix_time::time_duration, 1000000>')
    return pt::from_time_t(sec) + pt::microseconds(nsec/1000.0);
```

Adding an explicit cast to either int32_t or uint32_t seems to solve this issue.

My setup

- macOS 10.13.4
- clang Apple LLVM version 9.1.0
- boost 1.67.0

mikaelarguedas, 2018-06-06:
@dirk-thomas It looks like #79 addresses this issue

dirk-thomas, 2018-06-06:
Addressed by #79.

closed 2018-06-06 by dirk-thomas