ros/roscpp_core #91: improper implementation of ros::Time::useSystemTime()

sunnyroad, 2018-08-08:
https://github.com/ros/roscpp_core/blob/1110bb03024d5b698c481b41e5eeb7f4b5afbcbd/rostime/src/time.cpp#L285
may be should like this if we want to change to use SystemTime
```
void Time::useSystemTime()
{
  return g_use_sim_time = false;
}
```

dirk-thomas, 2018-08-09:
The function `bool Time::useSystemTime()` acts as a "getter" - not as a "setter" so it is fine as it is. (Maybe not named very intuitively.)

Assuming this resolves your "problem" I will go ahead and close the ticket. Please feel free to continue commenting on it.

closed 2018-08-09 by dirk-thomas

dirk-thomas, 2018-08-10:
The choice between system time and sim time is being made based on the [Clock](http://wiki.ros.org/Clock). So there is no API you expect to simply switch between them. When sim time is enabled and clock messages are received they set the time (https://github.com/ros/ros_comm/blob/6b9efd56d6882d1c017152ba11a5780a1496b30b/clients/roscpp/src/libros/init.cpp#L255-L258) which then changes the state of the clock (https://github.com/ros/roscpp_core/blob/1110bb03024d5b698c481b41e5eeb7f4b5afbcbd/rostime/src/time.cpp#L318-L324).