ros/roscpp_core #92: Durations not printing out properly with ROS_INFO, WARN, etc.

pbeeson, 2018-10-05:
Also filed issue here: https://github.com/ros/rosconsole/issues/20

I'm unsure if this ultimately is a rosconsole or a cpp_core issue.

```
double timer = 49.9999999999;
ROS_INFO_STREAM(timer<<" "<<ros::Duration(timer)<<" "<<ros::Duration(timer).toSec()<<" "<<ros::Duration(ros::Rate(1/timer))<<" "<<ros::Time(timer));
```

Output:
[ INFO] [1538739921.605671985]: 50 49.1000000000 50 50.000000000 50.000000000

pbeeson, 2018-10-05:
I believe the reason why this works for Time() but not for Duration() if that the fromSec() definition of Duration is not a rigorous as the one for Time in terms of handling rounding errors.

time.h:
```
    double toSec()  const { return (double)sec + 1e-9*(double)nsec; };
    T& fromSec(double t) {
      int64_t sec64 = (int64_t)floor(t);
      if (sec64 < 0 || sec64 > UINT_MAX)
        throw std::runtime_error("Time is out of dual 32-bit range");
      sec = (uint32_t)sec64;
      nsec = (uint32_t)boost::math::round((t-sec) * 1e9);
      // avoid rounding errors
      sec += (nsec / 1000000000ul);
      nsec %= 1000000000ul;
      return *static_cast<T*>(this);
    }
```

duration.h:
```
  template<class T>
  T& DurationBase<T>::fromSec(double d)
  {
    int64_t sec64 = (int64_t)floor(d);
    if (sec64 < INT_MIN || sec64 > INT_MAX)
      throw std::runtime_error("Duration is out of dual 32-bit range");
    sec = (int32_t)sec64;
    nsec = (int32_t)(nearbyint((d - (double)sec)*1000000000));
    return *static_cast<T*>(this);
  }
```

pbeeson, 2018-10-05:
Actually, what happens in the duration above is that the nearbyint() call on 0.9999999999*1e9 returns  1e9, which is 1 second, but then on printing the seconds are printed, then a ".", the nanoseconds are printed.  But there are too many nanoseconds.  The Time() function properly goes back and adds any nsec > 1e9 to the seconds column.

pbeeson, 2018-10-05:
The problem is that c++ decides that even though you request 
```  std::ostream& operator<<(std::ostream& os, const Time &rhs)
  {
    boost::io::ios_all_saver s(os);
    os << rhs.sec << "." << std::setw(9) << std::setfill('0') << rhs.nsec;
    return os;
  }
```

Additionally, in a fix for this, in both Time and Duration either the round() as in the Time should be used or the nearby() as in Duration should be used for consistency.  These won't return the same results by default for 0.5.

pbeeson, 2018-10-05:
Also, duration.h fromSec() just seems wrong in that durations can be negative, so the first floor() call on a negative double rounds down to the lower negative number.

dirk-thomas, 2019-01-26:
Addressed by #93.

closed 2019-01-26 by dirk-thomas