ros/roscpp_core #96: Misaligned address  for type 'double', in serialization.h, causing runtime error.

zhoulaifu, 2018-11-27:
I am testing ROS with ubsan, a sanitizer for detecting undefined behavior. Below is an issue ubsan found in `serialization.h`.  

    /opt/ros/kinetic/include/ros/serialization.h:234:5: runtime error: store to misaligned address 0x60600000aca4 for type 'double', which requires 8 byte alignment
    0x60600000aca4: note: pointer points here
      30 00 00 00 be be be be  be be be be be be be be  be be be be be be be be  be be be be be be be be
                  ^ 

 The issue was detected when I tested `turtlesim`.   To reproduce it, compile turtlesim with ubsan toggled, and then launch the node `turtlesim_node and you will see the message above. 

Many times misalignment is only a minor issue or even by design,  but I am not sure if it is the case here, given that the misalignment occurs in the core of ROS.

gavanderhoorn, 2018-11-28:
Can this be architecture specific?

zhoulaifu, 2018-11-28:
Possibly.  
```
$ uname -a
Linux ubuntu 4.15.0-39-generic #42~16.04.1-Ubuntu SMP Wed Oct 24 17:09:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

dirk-thomas, 2019-01-30:
@zhoulaifu The line number in your error message doesn't match the current code in the repository. Please try to reproduce the problem with the latest code and comment if you are still seeing the problem and if yes, include updated information.

zhoulaifu, 2019-02-05:
Which version of the repo and which distro would you want me to check? FYI, the error message above was from kinetic.

dirk-thomas, 2019-02-05:
As mentioned above: the line number in the error message doesn't match the code on the default branch so you might be using an older version. Please post the version number of the package you are using.

Please try to:
* update to the latest released Debian package or
* build the current HEAD of the default branch from source - that works either for Kinetic, (Lunar), or Melodic

zhoulaifu, 2019-02-12:
The issue disappears when I have updated the package `ros_kinetic_desktop_full` to the latest version. Thank you, @dirk-thomas.

closed 2019-02-12 by zhoulaifu