ros/roscpp_core #111: command-line error:invalid macro definition: =

Brabalawuka, 2019-07-16:
Generating cmake cache would result in command line error like this:

`E0992	command-line error: invalid macro definition: =`

It does not mention where this `=` is located but in the cpp code it locates at the very begining in line 1.
[image]

However I dont see any error on the command line preprocessor
```1> CMake generation started for configuration: 'x64-Debug'.
1> Command line: "cmd.exe" /c ""C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\PREVIEW\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja" -DCMAKE_INSTALL_PREFIX:PATH="C:\test\src\out\install\x64-Debug" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.22.27812/bin/HostX64/x64/cl.exe" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.22.27812/bin/HostX64/x64/cl.exe"  -DCMAKE_BUILD_TYPE="Debug" -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2019\PREVIEW\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\test\src" 2>&1"
1> Working directory: C:\test\src\out\build\x64-Debug```

dirk-thomas, 2019-07-17:
Since this happens on Windows I will add @kejxu to the ticket.

@Brabalawuka please include more information: what you are doing, exact steps, versions, environment, etc.

Brabalawuka, 2019-07-17:
I have solved this issue through a large number of configuration works and I will post how I solved it tmr morning i am in GMT+8

kejxu, 2019-07-17:
> I have solved this issue through a large number of configuration works

using Visual Studio for ROS projects would likely require proper configuration of include paths for most dependencies other than STL and platform apis (windows.h). Good to know you were able to get around with it!

Brabalawuka, 2019-07-18:
In the `CMakeSettings.json` file, do the following:

->>`"generator": "NMake Makefiles"` ->> This because `Catkin_make` uses `NMake` instead of `Ninja`  
    ->>`"configurationType": "RelWithDebInfo"` ->> Match with Catkin Type  
    ->>`"buildRoot": "C:\\catkin_ws\\build"` ->> Change build root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>`"installRoot": "C:\\catkin_ws\\install"` ->> Change install root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>`"cmakeCommandArgs": "DCATKIN_DEVEL_PREFIX=C:\\catkin_ws\\devel"` ->> Change develpment root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>~~`"buildCommandArgs": "-v"`~~  
    ->>`"cmakeExecutable": "C:/opt/rosdeps/x64/bin/cmake.exe" ` ->> Add this line to use ROS default cmake to avoid confusing errors.  

    Refer to following pictures:  

[image]  
[image]

These could probably sove those intellisense errors and can successfully compile and run.

closed 2019-07-18 by Brabalawuka