ros/roscpp_core #119: Integrating with OSS-Fuzz

Google-Autofuzz, 2020-04-13:
Greetings roscpp_core developers and contributors,

We’re reaching out because your project is an important part of the open source ecosystem, and we’d like to invite you to integrate with our [fuzzing](https://www.owasp.org/index.php/Fuzzing) service, [OSS-Fuzz]( https://opensource.googleblog.com/2016/12/announcing-oss-fuzz-continuous-fuzzing.html ). OSS-Fuzz is a free fuzzing infrastructure you can use to identify security vulnerabilities and stability bugs in your project. OSS-Fuzz will:

- Continuously run at scale all the fuzzers you write.
- Alert you when it finds issues.
- Automatically close issues after they’ve been fixed by a commit.

Many widely used [open source projects]( https://github.com/google/oss-fuzz/tree/master/projects ) like OpenSSL, FFmpeg, LibreOffice, and ImageMagick are fuzzing via OSS-Fuzz, which helps them find and remediate [critical issues]( https://bugs.chromium.org/p/oss-fuzz/issues/list?can=1&q=status%3AFixed%2CVerified+Type%3ABug%2CBug-Security+-component%3AInfra+ ). 

Even though typical integrations can be done in < 100 LoC, we have a [reward program]( https://www.google.com/about/appsecurity/patch-rewards/ ) in place which aims to recognize folks who are not just contributing to open source, but are also working hard to make it more secure.

We want to stress that anyone who meets the eligibility criteria and integrates a project with OSS-Fuzz is eligible for a reward.

If you're not interested in integrating with OSS-Fuzz, it would be helpful for us to understand why—lack of interest, lack of time, or something else—so we can better support projects like yours in the future.

If we’ve missed your question in our [FAQ]( https://google.github.io/oss-fuzz/faq/ ), feel free to reply or reach out to us at oss-fuzz-outreach@googlegroups.com.

Thanks!

Tommy
OSS-Fuzz Team

dirk-thomas, 2020-04-13:
Please feel free to contribute pull requests to use your fuzzing service. It is unlikely that the maintainer will have the resources to do so.

Google-Autofuzz, 2020-04-14:
For sure, we can help upstream roscpp_core to OSS-Fuzz as a start. We just need a maintainer email address for the bugs to be sent to if OSS-Fuzz finds any. @dirk-thomas

dirk-thomas, 2020-04-14:
Maybe start with a first run and post the results to this ticket?

Google-Autofuzz, 2020-04-14:
It will be best to provide an email address because in this way you get access to the crash reports and fuzzer statistics. This is explained here: https://google.github.io/oss-fuzz/getting-started/new-project-guide/#primary 

It is a requirement to have an email address for roscpp_core to be upstream to [OSS-Fuzz](https://google.github.io/oss-fuzz/getting-started/new-project-guide/).

closed 2020-06-11 by Google-Autofuzz