ros/roscpp_core #120: ros::Time and ros::Duration don't respect TIME_MAX and DURATION_MAX in arethmetic operators

fabmene, 2020-04-17:
roscpp_core/rostime/src/time.cpp defines `TIME_MIN`, `TIME_MAX`, `DURATION_MIN` and `DURATION_MAX` with `std::numeric_limits`. Performing arithmetic operations on them throws an exception due to causing an overflow.

From a user's standpoint however, I would expect
`ros::Time::now() + ros::DURATION_MAX == ros::TIME_MAX`

Is there a design reason why this is not implemented?

dirk-thomas, 2020-04-17:
> From a user's standpoint however, I would expect
> `ros::Time::now() + ros::DURATION_MAX == ros::TIME_MAX`

I am not sure I share this expectation. For now() being zero I might agree but for anything else I would consider this to be an overflow which an exception seems appropriate for.

fabmene, 2020-04-20:
Maybe I misunderstand the use of these constants. I intended to use DURATION_MAX as resembling an infinite duration and understood TIME_MAX to represent a point of time infinitely far into the future.

Basically I expected a representation for an infinite duration that works with the `ros::Time` and `ros::Duration` arithmetics and doesn't require me to check for overflows in the process.

dirk-thomas, 2020-04-20:
As other constants ending in `_MAX` these values describe the largest value which can be stored in the data type. They certainly do not represent infinity.

So in order to check overflows you have to catch exceptions.

Assuming this answered your question I will go ahead and close this ticket.

closed 2020-04-20 by dirk-thomas