ros/roscpp_core #127: ros::Rate::sleep is not working as expected when the rate is larger than the publish rate of /clock

timonegk, 2021-06-11:
We are currently adapting much of our software stack to work in simulation. There, `/clock` is published with a rate of 125Hz. In some parts of our software, we are using ros rates at higher rates (for example 500Hz). In these cases, `rate.sleep()` does not work as expected (blocking until the time progresses) and instead returns immediately. This leads to those parts of our software running at unlimited rates, using too much resources.
In my opinion, in this case, `rate.sleep()` should block like it does in python.

timonegk, 2021-06-11:
`rate.sleep()` does return `false` in that case, so it would be possible to listen for that and sleep manually then. But the purpose of returning `false` seems to be for backwards jumps in time, so I don't know if that is a reliable way of circumventing the problem.

peci1, 2025-05-04:
Closing this issue as invalid.

closed 2025-05-04 by peci1