ros/roscpp_core #130: ros serialization get wrong message size when compile on arm64 with gcc -O3

randoms, 2021-11-19:
Example Project:  https://github.com/AutoxingTech/simple_publisher_crash

output message size

```
Publication::enqueueMessage 1 2427156042
Publication::enqueueMessage 2 2427156042
Publication::enqueueMessage 3
```

randoms, 2021-11-19:
This is a bug of gcc. The issue was caused by `-ftree-loop-vectorize` optimization option which is included by -O3 option. The issue was reported [here](https://gcc.gnu.org/bugzilla//show_bug.cgi?id=102435) and still not fixed in gcc 9.3. Upgrade to gcc 10.x should fix this issue.

peci1, 2023-04-09:
Would it be a solution to mark some function in the serialization code as non-optimized, or e.g. `-O2`-optimized? https://stackoverflow.com/a/9044490/1076564 . Can you try?

peci1, 2023-06-06:
@randoms #136 should fix this. Could you verify?

PR #136 (2023-06-07): Fixed wrong serialization of messages on ARM with GCC 9.3 and -O3 optimizations
Closes #130.

@randoms Do you have a way to verify this fix helps?

I was able to reproduce the bug on arm64 Jetson with GCC 9.3.0. When I tested with 9.4.0 (which is now the default in Focal), the bug did not manifest. So I limited the fix to the only known combination triggering this issue - arm64 + GCC 9.3.
commit 3ab19cd4 (2023-06-14): Fixed wrong serialization of messages on ARM with GCC 9.3 and -O3 optimizations

Closes #130.

(cherry picked from commit 5073e6b8071524174c13a3acb988affe6ff49898)
closed 2023-06-07 by peci1