ros/roscpp_core #131: inclusion of unecessary boost from ros console and ros time

thraen, 2021-12-23:
Hi, 
We face long build times.
In many compilation units `ros/console.h` is all that is is needed from ros. But that includes a lot of unecessary boost:
ros/console.h includes ros/time
ros/time.h includes boost/special_functions/round 
boost/special_functions/round in turn includes more of boost than needed for rounding.

But 'fixing' this in ros/time, by using other means of rounding, breaks ros/console, because ros/console also needs boost shared ptr, that implicitely came with the special functions. 
Fixing this in ros/console, eg by using std::shared ptr, would probably lead to similar problems in depending projects.

Are there plans for gradually removing boost from the core of ros? What can we do to help?

BR!

mikepurvis, 2022-01-24:
I believe the path forward on this is migrating to ROS 2, which was built from the start against C++11 so that boost would not be a necessary dependency, at least for the core libraries.

peci1, 2023-06-06:
Taking into account Melodic is almost EOL and Noetic cannot receive any API breaking changes, this can't be done. The includes in public header files are public API of the packages.

Closing this issue.

closed 2023-06-06 by peci1