ros/roscpp_core #133: Resolution of ros::Duration::sleep() limited by implicit sleep in nanoseconds

YoshuaNava, 2022-03-21:
Hi,
By reading through the code of `rostime` I found that ros::Duration::sleep() seems to have an internal sleep in wall time with a length of 1,000,000 nanoseconds = 1 millisecond. This is the exact line of code where that happens -> https://github.com/ros/roscpp_core/blob/noetic-devel/rostime/src/time.cpp#L402

As far as I understand, the call to `ros_wallsleep(0, 1000000);` might be limiting the maximum resolution of sleep to 1millisecond. Therefore, if someone inside an application wants to sleep for e.g. 0.5ms, that wouldn't be possible. Is this right? Can some of the authors/maintainers confirm this behavior?

If so, would you be open to a fix/extension to allow for more fine-grained sleep durations? I intend to use ros::Time inside a node that spins at a rate >=400Hz (every 2.5ms), and this aspect of the current implementation would likely increase the probability of missing processing deadlines, because with the current implementation sleep would wake up at t=2ms, or t=3ms, but never t=2.5ms.

Thank you very much in advance.

Best regards,
Yoshua Nava

MatthijsBurgh, 2023-01-31:
When running at such a high frequency of 400Hz, I would think WallDuration/WallTime is better.

YoshuaNava, 2023-01-31:
Hi @MatthijsBurgh,

We have tried your suggestion. However, running in WallTime means that any high-rate module will not listen to the updates of ROS Time. Thus, if you slow down, speed up, or pause

* The replaying of a ROS bag,
* Simulation,

The nodes notion of time will be different from that of the sensor and actuator clocks, leading to ambiguous situations. Furthermore, if you have Time-based events, they will be triggered at the wrong instant. Breaking away into a different time domain can increase the number of issues, but not solve the problem at the root.

To give a practical example: imagine if you have a swarm-drone system, that relies on accurate, time-sensitive motion planning. If you want to test this system in simulation, with WallTime, you need to ensure that your simulator will run at exactly the same rate as your host PC Clock, otherwise your whole motion planning algorithm will have to 'fight' uncertain timing (which is a problem that you can solve in the real world with a GPS)

Hence, that's why my suggestion goes into the direction of making the **minimum wait time** an optional argument.

WDYT?

Best,
Yoshua

MatthijsBurgh, 2023-01-31:
I agree with you. The WallTime doesn't slow down. I expected your process to be time critical, because of the high frequency.

I think your case with such a high frequency, but want to sync with the ROS time, is very rare. And as ROS1 is going to the end of its life, I don't expect the maintainers to change it.

YoshuaNava, 2023-01-31:
What you mention makes sense :+1:.  I hope that this issue's info can be useful for future travelers of the ROS1 code, or authors of ROS2.

Thank you for the feedback @MatthijsBurgh :muscle:

peci1, 2023-04-09:
I can imagine a PR along these lines could get merged: compare the intended sleep duration with the `1 ms` sleep, and if one tenth of the intended duration is shorter than the `1 ms`, shorten the wallsleep to this one tenth. This should not affect most normal use-cases, but should help your cause.

peci1, 2023-06-06:
@YoshuaNava #137 should help with this. Could you verify?

closed 2023-06-13 by peci1