ros/roscpp_core #146: ROS Noetic end-of-life plan?

sloretz, 2025-05-02:
Hello @peci1 

I see this repo contains packages in [desktop_full](https://index.ros.org/p/desktop_full/#noetic), and you are the listed maintainer. What is your plan for [ROS Noetic end-of-life on May 31st](https://discourse.ros.org/t/ros-noetic-end-of-life-may-31-2025/43160)?

FYI starting **May 16th** ROS Noetic releases will only be accepted into [ros/rosdistro](https://github.com/ros/rosdistro) if they fix regressions, meaning releases will be required to fix something that was working previously.

---

If you don't have a plan yet, I recommend:

* Closing all pull requests breaking API or ABI
* Closing all pull requests adding new features
* Asking bug fix PR authors to recruit multiple reviewers to test it

If your repository only supports ROS 1, and you are not going to maintain it after ROS Noetic goes EOL, then make that clear to your users on May 31, 2025 by:

* Closing open PRs and issues
* Archiving the repo

Here are saved replies I've been using:

## Bug fix PR

```
Thank you for the PR!

[ROS Noetic will reach end-of-life on May 31st, 2025](https://discourse.ros.org/t/ros-noetic-end-of-life-may-31-2025/43160).
Every change comes with a risk of introducing regressions, and there isn't much time left to fix them.
To make sure this PR doesn't introduce any regressions please:

* Describe how you tested this change
* Recruit at least one more person to review this PR and try it out on their system
```

## Feature PR

```
Thank you for the PR!

Unfortunately I don't think we should merge this one.

[ROS Noetic will reach end-of-life on May 31st, 2025](https://discourse.ros.org/t/ros-noetic-end-of-life-may-31-2025/43160).
Every change comes with a risk of introducing regressions, and there isn't much time left to fix them.
I'm closing pull requests that add features so that the remaining time is allocated towards bug fixes and compatibility with newer Ubuntu distros.
```

peci1, 2025-05-02:
Oh, right, I haven't figured this one's on me :) I'll have a look at it over the weekend.

peci1, 2025-05-04:
I closed the PRs and issues.

The plan is that I won't archive this repo, making it possible to use it and continue further development for ROS-O if there is interest. I've added a note about it in https://github.com/ros/roscpp_core/commit/da2394dd175c9d2d39b9756a33d35a7ae2f79d12 .

closed 2025-05-04 by peci1